atuline/WLED #48: What's the protocol for audio? Windows / Python App? as source?

zombodotcom, 2020-11-26:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Thank you for your ideas for making WLED better!

THATDONFC, 2021-01-05:
Please update your original post to include the information necessary for us to help you.

atuline, 2021-01-05:
I'm going to assume he's talking about the UDP sound sync, which is documented at:

https://github.com/atuline/WLED/wiki/UDP-Sound-Sync

closed 2021-01-05 by atuline