atuline/WLED #49: Error 10: could not mount filesystem

emdeex, 2020-12-09:
**Describe the bug**
I'm using the 11.0 sound reactive binary, it's flashed OK and the WLED-AP comes up... but in the UI I get two errors flashing up;

Error 10: could not mount filesystem!
SyntaxError: UnexpectedToken in JSON at position <0

**To Reproduce**
Just flashed the binary.

**Expected behavior**
No errors

**WLED version**
 - Board: ESP32 WROOM
 - Version 11
 - Format binary

**Additional context**

Aircoookie, 2020-12-10:
Hi! Please connect to the AP, go to security settings, check "Factory reset" and save. It will format the filesystem, after that the error should go away :)
If it persists, your ESP32 might have a flash size smaller than 4Mb, in which case a custom compilation might be neeeded.

emdeex, 2020-12-10:
> Hi! Please connect to the AP, go to security settings, check "Factory reset" and save. It will format the filesystem, after that the error should go away :)
> If it persists, your ESP32 might have a flash size smaller than 4Mb, in which case a custom compilation might be neeeded.

Thanks!  I've given that a go... it seems to work... I go back to the AP mode, and set the WiFi settings... connect back to the new IP... and get the same errors coming up.  I repeated the factory reset, same result.

This is a ESP32 WROOM with 4MB.

emdeex, 2020-12-12:
@Aircoookie Do you have any other ideas I can try?  It seems like its an issue with creating the new filesystem right?

emdeex, 2020-12-13:
I decided to try VS Code and Platform IO as the IDE.... I installed it, grabbed the git, set the board..... compiled it and flashed it.... no errors anymore.

So whatever was going wrong was somewhere with Arduino IDE... not sure what..... but I'll close this issue now as it doesn't seem to be a code issue.

closed 2020-12-13 by emdeex