atuline/WLED #50: Lights freezes on OLED display refresh

zsolesz79, 2020-12-13:
**Describe the bug**
When using the **ssd1306_i2c_oled_u8g2 usermod** the lights freezes for a couple of millisecs every time the OLED display refreshes. Tried with usermod type v1 and v2 too. The refresh rate of the OLED display is also very slow (compared to the non soundreactive version of WLED)

**To Reproduce**
Change effect or change color palette.

**Expected behavior**
No hangs and hickups in the light show.

**WLED version**
 - Board: ESP32 dev and ESP32 Mini
 - Version: 0.11.0, latest available source today
 - Format: self-compiled

Thank you for your help!

THATDONFC, 2021-01-05:
Hmmmm I'm not sure where to start with this one. My assumption would be there is something interrupting the led data when the screen is updated. What are you displaying on the screen?

zsolesz79, 2021-01-06:
I'm using the code found under /usermods/ssd1306_i2c_oled_u8g2/  It shows:

    Current SSID
    IP address if obtained
        in AP mode and turned off lightning AP password is shown
    Current effect
    Current palette
    On/Off icon (sun/moon)

I have no problem with the original WLED version just with SR fork.

softhack007, 2022-05-06:
it had a similar "slow down" experience once, with usermod_v2_four_line_display and `U8X8_SH1106_128X64_...` oled I2C display. Here is how I solved it.

Background: On ESP32 you can assign any PIN to have i2C an "hardware" bus. I picked `FLD_PIN_SCL=4` and `FLD_PIN_SDA=0`. As it turned out, this was the cause for slow display updates. The usermod was concluding that i did not use the standard PINs for I2C, so it picked a "software" driver.

https://github.com/atuline/WLED/blob/3a4e85058fb6c3e2f59ee06be2cc527b1f421f68/usermods/usermod_v2_four_line_display/usermod_v2_four_line_display.h#L28-L30

https://github.com/atuline/WLED/blob/3a4e85058fb6c3e2f59ee06be2cc527b1f421f68/usermods/usermod_v2_four_line_display/usermod_v2_four_line_display.h#L226

https://github.com/atuline/WLED/blob/3a4e85058fb6c3e2f59ee06be2cc527b1f421f68/usermods/usermod_v2_four_line_display/usermod_v2_four_line_display.h#L239-L241

.

**Long story short**: make sure that the usermod runs with hardware I2C. 
Two options:
a) use PIN21 and PIN22, as these are the "default" I2C PINs, recognized as hardware.
b) modify the usermod source code (line 29+30) so that HW_PIN_SCL and HW_PIN_SDA are in line with your wiring.

Hope this helps. In general, on ESP32 any PIN does support hardware I2C, so it would not be necessary to fall back to software I2C which is of course slower.

softhack007, 2022-07-08:
Problem in upstream WLED. See https://github.com/Aircoookie/WLED/issues/2708 .

closed 2022-07-08 by softhack007