atuline/WLED #55: mobile phone buil-in microphone

mehranviva, 2020-12-25:
hi thanks for your good job with audio reactive

THATDONFC, 2021-01-05:
Sorry, this is not in the scope of this project. If you come up with a solution, you are welcome to submit a pull request and we can revisit this at that time.

closed 2021-01-05 by THATDONFC

jterrell345, 2022-11-19:
A suggestion on how to do this is to use the WLED Phone APP. The App could be enhanced to pick up surrounding audio and then send the beat to the Microprocessor.

softhack007, 2022-11-19:
Actually there would be a lot of work to do this.
The WLED app (https://github.com/Aircoookie/WLED-App) is basically a wrapper around a web browser that shows the WLED UI.

Phone apps are a completely different technology, and I have not the slightest idea how to make a phone app that can transfer audio data in teatime to the ESP32.

If you want to try yourself and come up with a solution, I guess @aircoookie will be happy to receive a PR (pull request). 

It might be possible to write a phone app that uses UDP to transfer our "sound sync" format to WLED. 

However there are still no plans to make such an app in the frame of SR WLED, sorry.