atuline/WLED #56: Direct Connect Stereo Headphone / mono Speaker

bluenazgul, 2020-12-27:
Would it be possible to Direct Connect Stereo Headphone / mono Speaker to A0 on ESP8266?

In my case i dont realy have room to use an additional Microphone.

THATDONFC, 2021-01-05:
I'm not exactly sure what your question is. Have a look at our wiki for information on analog and digital audio input options.

atuline, 2021-01-05:
I would not recommend connecting directly to A0, because it's a 3.3V device and the signal should be centered around 1.6V.

Our Wiki shows how to connect a line-in signal:

https://github.com/atuline/WLED/wiki/Analog-Audio-Input-Options

closed 2021-01-05 by atuline

estevez-dev, 2021-01-11:
Hi @atuline 
Almost everything is clear in the wiki pages but I can't understand where 3.3v goes on [Line In Wiring Example](https://github.com/atuline/WLED/wiki/Analog-Audio-Input-Options#the-following-schematics-are-provided-as-an-example-only-there-are-many-ways-to-achieve-the-same-results-these-are-only-a-few-of-those-ways). Should it be connected to `VIN3.3V` pin on ESP32 board or I should provide 3.3V power input on that end?