atuline/WLED #61: INMP441 how conect to D1Mini?

salopette, 2021-01-10:
Hello, thanks for the WLED Mod!
I have an INMP441 microphone, which pins do I have to connect to the D1 Mini?

The micro has 6 (l / r, WS, SCK, GND, VDD and SD)
Thanks for your help!

atuline, 2021-01-10:
If it's an ESP8266, you don't. If it's an ESP32, you may need to re-define the pins. Check the Discord for what others have done.

closed 2021-01-10 by atuline

salopette, 2021-01-10:
I have a Wemos D1 Mini. https://www.wemos.cc/en/latest/d1/d1_mini.html
So doesn't it work with that?

atuline, 2021-01-10:
That page clearly says it's an ESP8266 (and I have about 30 WeMOS D1 Mini's), so no it doesn't support the INMP441. And the reason is that ESP32 supports I2S, while ESP8266 doesn't.

salopette, 2021-01-10:
Thanks, now I understand it!
Is there any other microphone that is supported by esp8266 with the bin file?
I don't want to swap my 10 D1 for esp32 now.

atuline, 2021-01-10:
https://github.com/atuline/WLED/wiki/Analog-Audio-Input-Options

salopette, 2021-01-10:
Thanks a lot!