atuline/WLED #65: MQTT does not save

jappyjan, 2021-01-25:
**Describe the bug**
When entering MQTT Settings (and checking the activation checkbox) those settings wont be saved when clicking on save.

**To Reproduce**
Lolin32 Lite with latest atuline/wled release flashed. Mqtt settings entered. 

**Expected behavior**
mqtt settings shall be saved

**WLED version**
 - Board: lolin32 lite
 - Format binary 0.11.1

Thank you for your help!

jappyjan, 2021-01-26:
i was able to save MQTT by compiling the code myself and remove the "WLED_DISABLE_MQTT" flag from the platformio.ini env.

why did you guys disable it in the first place?

atuline, 2021-01-26:
Originally for the ESP8266, which was VERY limited in available resources. We also had reliability issues with the ESP8266, so we disabled everything that was not tied down.

closed 2021-01-26 by atuline

jappyjan, 2021-01-26:
I get that, but it was also proactively disabled for esp32dev... maybe remove the flag in the future as it definitely works with mqtt enabled on esp32 chips with no problems at all

jappyjan, 2021-01-26:
And thank you for your great work! I really enjoy it!

atuline, 2021-01-26:
Thanks. With the challenges we've had, we felt consistency was the way to go.