atuline/WLED #66: error afther uploading firmware

mich1974, 2021-01-28:
Hi, 

when I upload soundReactive_WLED_0.11.1_ESP32_ledpin16.bin to my huzzah32 with microphone amplifire Max4466 I resieve following error
Showing logs:
[01:06:00] .pio / libdeps / soundReactive_esp32_LEDPIN_16 / LittleFS_esp32 / src / lfs.c: 1003: error: Corrupted dir pair at {0x0, 0x1}
[01:06:00] E (59) esp_littlefs: mount failed, (-84)
[01:06:00] E (62) esp_littlefs: Failed to initialize LittleFS
[01:06:01] I2S driver installed.
[01:06:01] Digital microphone is NOT present.
[01:06:01] Ada (edited)

WLED version**
 - Board: Adafruit HUZZAH32 - ESP32 Feather
 - Format: soundReactive_WLED_0.11.1_ESP32_ledpin16.bin
 - microphone amplifire Max4466
Used pins
Grd to ledstrip SK1286, power usb to ledstrip datapin Rx or gpo16 (this is working with regular wled version11 bin file WLED_0.11.1_ESP32_ledpin16
Grd to microphone amplifire Max4466 power 3.3v and pin A1or GPO36 for data.

Thank you for your help!

madmic1314, 2021-03-13:
I also have the same issue I believe - after flashing precompiled binaries with esphome the first boot gives the same error when monitoring the serial line. Reboot the device and it works perfectly.

Power externally (USB powerbank or 10A) and get the same thing, seems to not detect the I2C mic (INMP441 in my case).

Seems everything boots ok if the serial port is in use?

madmic1314, 2021-03-13:
Flashed pin16 version and the same result.

However I've noticed that after powering externally or with the serial line off if I press the EN button everything boots ok and works perfectly - this explains serial monitor as this causes a reboot. Seems to be an initialisation / boot timing bug at initial power on?

I've ordered a different brand to test on this. But for now I can simply put a push button across the EN pin & ground to do the same thing.

Anything else we can do to debug this?

isabelgodfrey, 2021-12-25:
Even I too had the same problem while following this [tutorial](https://www.circuitschools.com/diy-smart-led-strip-with-sound-reactive-effects-using-esp8266-esp32-wifi/). but When i removed data pin from the board and uploaded the binary file it is uploaded successfully. but im not able to find the pins for connecting microphone as my microphone module has 4 pins. can anyone share the wiring image.

softhack007, 2022-07-14:
The logs that you show suggest that it's actually a problem related to flash filesystem (LittleFS).
Please note that flashing with esphome tool is not supported any more:

https://github.com/atuline/WLED/wiki/Installing-and-Compiling#flashing-esp32-binaries-with-esptool
>> Warning: We had to change the partition size on the ESP32 in order to 'fit' all the new features. This means that the 'old way' of upgrading/flashing, no longer work unfortunately. You cannot use ESPHome Flasher, and you cannot use OTA from a build prior to b5.

Closing this issue report, as it relates to a very old release, and there was not activity for a longer time.

closed 2022-07-14 by softhack007