atuline/WLED #67: Enabling Audio Sync - Crashes Network.

michaeljvdh, 2021-02-09:
**Describe the bug**
When enabling the feature "Transmit" crashes the network. (potentially flooding), if I disconnect the unit it stops .. connect it again and keeps flooding, brings down everyone on my network.

**To Reproduce**
Tried changing the Transmit port, same issue even if port 10000 steps apart.

**Expected behavior**
Transmit should save without crashing the network (probably flooding) and the receiving unit should act accordingly.

**WLED version**
 - ESP32
 - 11.1
 - .bin

**Additional context**
Anything else you'd like to say about the problem?

Thank you for your help!

atuline, 2021-02-09:
According to line 108 of usermod.cpp in both our master and dev branches, we are transmitting a UDP packet every 20ms by using the FastLED EVERY_N_MILLIS() function. Unless that function is broken, 50 packets per second doesn't seem to be a lot, even over 2.4G.

See: https://github.com/atuline/WLED/blob/2cf6b5f2388e2f99a469e71ecfefac26f0bc3e9e/wled00/usermod.cpp#L109

michaeljvdh, 2021-02-10:
I understand, but literally when I clicked save my whole network shutdown.  Disconnect reconnect it happens again, refreshed 4 times .. same result.  Thank though.  Great project.  PS: just to be clear the unit is on DHCP so I have no conflicts.

atuline, 2021-02-14:
I tried to replicate this issue with https://github.com/atuline/WLED/tree/dev

Ran a WiFi analyzer as well as Speedtest. Could not replicate this issue. In my case, I use static IP addresses because I have several devices and need to know which is which. I also confirmed that the Audio sync was set to Transmit on UDP port 11988.

yfford, 2021-02-15:
I have suffered this exact same issue. I have been running an ESP32 to sync the sound out to other ESP32's, when i was running on 11.0 i had no issues but as soon i as i updated to 11.1 the same ESP32 seems to flood the network taking down the WIFI. as soon as i disconnect the ESP32 that is transmitting the WIFI comes straight back.
I have an INMP441 mic connected to the ESP if i remove the power from the mic it also removes the WIFI issue.
I have the gain settings on 0 and squelch on 3 if i move the gain any higher i start to have WIFI issues straight away.

atuline, 2021-02-15:
OK, guys, I'd need more information, such as the specific link you used to download the software, the board(s) you are using that cause the issue (I only have a LOLIN D32), your microphone configuration, your WiFi configuration, i.e. AP mode, static IP, DHCP, and most importantly, what tool you're using to demonstrably prove that your network is 'going down'. Oh and specific steps/configuration I can perform in order to attempt to re-create this would be helpful, and does this happen on bootup, or do you needs to run an animation.

yfford, 2021-02-15:
ok here we go as much info as i can give.
I have only ever used the code from the links here.  i use the PIN16 version
I have several different supplies of ESP32
I have 
AZDelivery LOLIN32 Lite V1.0 ESP-32 Rev1 WiFi Bluetooth Board 4MB Flash 500mA CH340 CH340G Lithium Battery Connector ESP32
AITRIP 3 Pack for ESP32-DevKitC core Board ESP32 Development Board ESP32-WROOM-32D for Arduino IDE (ESP32-WROOM-32D)
i have tried both of the above and have had the same issue with both.
I use 1 ESP32 to feed sound to 5 other ESP32's.
I am using INMP441 I2S digital microphone connected as per the information on this Wiki. Gain 0 Squelch 3, the Mic works well and lights show no animations when no sound detected.
I have a brand new Ubiquti Lite access point and an Ubiquiti edge x running isolated it only has the wifi dish connected. The Edge X is providing DHCP for the ESP's.
All ESP's are on DHCP currently (I have had them connected via static made no difference).
Also before the Ubiquti AP was purchased i was using a netgear access point and the experienced the same issue with the wifi.
The issue i see with the Wifi is as follows i try to connect to my nasty2G ssid on my ipad to allow it to talk to WLED ipad app and as the IPAD connects it says incorrect password and refuses to connect i saw this on the ubiquiti ap and the netgear AP if i unplug the transmitting ESP32 i connect to the nasty2G ssid without an issue. Also my nest thermostat gets disconnected (this is on an entirely different network and access point, again if i disconnect the transmitting ESP32 the nest reconnects on its own)
It does not happen on boot but sometimes it happens really quick and other times it takes a bit longer, i do run animation on the Main ESP32 mostly **pixelwave, somtimes i sync the animantion to the other strips but have seen this when i am using different animations on the other strips.

To me it just seems to flood the WIFI and kills access to the AP,

I have been trying to get this working correctly i have purchased 4 digital mics, at least 3 different brands of esp32, i have used 2 different access points and have a virgin router for DHCP and network, and also Ubiquti edge x for DHCP, and every combination of that kit :)

atuline, 2021-02-15:
OK, thanks. I didn't see a link there, so I'll assume you flashed a binary as opposed to compiling/uploading from source. I'm also going to assume it was this binary: https://github.com/atuline/WLED/releases/download/v0.11.1/soundReactive_WLED_0.11.1_ESP32_ledpin16.bin

yfford, 2021-02-15:
Yes sorry i flashed the binary you linked above.

michaeljvdh, 2021-02-15:
AH THANK HEAVENS thought I was going mad.

Only difference with me is is used esp32 standard 11.1 and with Esp32wroom and max9814 mic

I used the .bin file

atuline, 2021-02-15:
Interestingly, I flashed the ESP32.bin file using NodeMCU-PyFlasher 4.0-x64.exe to my LOLIN D32 board and was getting a "flash read err, 1000", and it wasn't running for me at all. Same with the 16 pin version.  On the other hand, I was able to compile/upload from source and everything ran. Another phone was running YouTube on that WiFi network with no issues. I'll ask @THATDONFC to see if he can re-create the release from recent source.

yfford, 2021-02-16:
Thats awesome at least i can stop buying stuff now :) i do use ESP home flasher if that makes any difference...thanks for looking into this i love this project....

yfford, 2021-02-16:
I have managed to compile and upload from source. it might just be me but i have been staring at these lights for what seems like months now and with this complied source everything seems a bit snappier and dare i say more stable than i am used to. I have the master syncing out to 5 separate strips and everything is looking good so far, i will update in a couple of hours but its looking promising, thank you for the help with this.

michaeljvdh, 2021-02-16:
@atuline Thank you very much for the effort.  @yfford thanks for joining in and saving my sanity lol.  Ill leave this open until I test mine too.  Should be today.

atuline, 2021-02-16:
Spending a bit more time on this, and we're going to be getting the other team members in to have a closer look. Even on the master/dev source branches, I've found that ping response to the ESP is ~200ms when UDP transmit is disabled and ~5ms when it's enabled.

@yfford - In the meantime, one does not simply 'stop buying stuff'.

yfford, 2021-02-16:
@atuline You are quite right, just bought you a coffee for all the hard work....cheers mate..5 hours now and things are looking good....
@michaeljvdh i would not have commented if i had not seen your post i thought it was just me.

atuline, 2021-02-16:
Oh, and gentlemen, you might also want to go to the WiFi settings page and check 'Disable WiFi sleep'. That's been problematic, and if that solves your .bin issue, I'll make sure it's well documented on our Wiki.

yfford, 2021-02-16:
I did see that and tried it before didnt make a difference but i will do it again now

THATDONFC, 2021-02-16:
I see an issue with udp sound sync transmit flooding UDP 11988. I'll have to take a really close look at this to see why it might be happening. My advice for now, if this is affecting your wireless network, disable UDP sound sync for now. Technically a modern router should not allow this type of thing to bring down a network. 

@yfford do you know how to run a tcpdump on you UI AP? If so, could you run one on the 2.4 interface and let me know what you see?

yfford, 2021-02-16:
@THATDONFC i have done a bit of googlefu is this what you want? if so how much?
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on eth0, link-type EN10MB (Ethernet), capture size 262144 bytes
21:48:57.810153 IP 192.168.10.38.60893 > 10.0.0.51.22: Flags [.], ack 3779086908, win 8210, length 0
21:48:57.810963 IP 10.0.0.51.22 > 192.168.10.38.60893: Flags [P.], seq 1:117, ack 0, win 2300, length 116
21:48:57.811367 IP 10.0.0.51.22 > 192.168.10.38.60893: Flags [P.], seq 117:233, ack 0, win 2300, length 116
21:48:57.811796 IP 192.168.10.38.60893 > 10.0.0.51.22: Flags [.], ack 233, win 8209, length 0
21:48:57.812070 IP 10.0.0.51.40561 > 10.0.0.1.53: 20679+ PTR? 38.10.168.192.in-addr.arpa. (44)
21:48:57.812442 IP 10.0.0.1.53 > 10.0.0.51.40561: 20679 Refused 0/0/0 (44)
21:48:57.826574 IP 10.0.0.54.11988 > 239.0.0.1.11988: UDP, length 88
21:49:07.818492 IP 10.0.0.51.36730 > 10.0.0.1.53: 42630+ PTR? 1.0.0.10.in-addr.arpa. (39)
21:49:07.818867 IP 10.0.0.51.22 > 192.168.10.38.60893: Flags [P.], seq 233:685, ack 0, win 2300, length 452
21:49:07.819077 IP 10.0.0.1.53 > 10.0.0.51.36730: 42630 Refused 0/0/0 (39)
21:49:07.823611 IP 10.0.0.54.11988 > 239.0.0.1.11988: UDP, length 88

ewoudwijma, 2022-07-15:
Closed as no activity in this issue

closed 2022-07-15 by ewoudwijma