atuline/WLED #68: volume controlled brightness

ewoudwijma, 2021-03-08:
Hi Atuline,

I think it might be an interesting idea to use volume to control brightness of an effect, for example the spectral effect could benefit from it. Is this something considered before? In that case volume controlled brightness would be a checkbox per effect or a general setting.
In case this is complex, could you add 'spectral-by-volume' as an extra effect? I guess this will become my favorite ;-)

atuline, 2021-03-08:
Hi, in the past I've had several effects that used volume to control the brightness. The challenge with WLED is that by supporting so many different inputs (MAX4466, INMP401, MAX9814, INMP441, ICS-43434 and line-in) with so many different signal levels that some would barely affect the led's while others would be too much. As a result, I typically amplified the value to be bright and blaring.  If you look at FX.cpp, for the volume only routines (i.e. a single '*'), there's a variable called 'sampleAgc'. If you replaced that with a variable called 'sample' you'll get more of a volume brightness effect. . . with a result that depends on your microphone, the current loudness and the brightness of the room. We're hoping that AirCoookie adds more modular controls to WLED that we may be able to turn on/off so that we could have a 'sampleAgc/sample' checkbox. Kind of hesitant to add more controls and further deviate from WLED's architecture for now.

ewoudwijma, 2021-03-09:
Thanks for your response. Agree with you that we should stay aligned with WLED as much as possible. I read somewhere that aircookie is considering to have controls per effect. That would help a lot I guess. Regarding the different audio inputs, I would have thought the gain setting is leveling this out. I currently use pre builds but hope to find the time in the future to compile and play with settings myself.

atuline, 2021-03-09:
Yea, the gain settings help to a degree. I may experiment with something like brightness = sample*slider  (where slider has been mapped) or something along those lines. Currently looking to see if I can improve the beat detection.

ewoudwijma, 2021-03-10:
Haha 'sounds' good! I will give it a try! Beat detection is currently not in the builds isn't it? I think that has higher priority 😉. 
One more suggestion I have is: WLED has the option to cycle through the presets. What you can do with sound is that you switch presets after a silence. Is that difficult to implement?

atuline, 2021-03-12:
Oh, and I removed spectral a while back, as it was written for a fixed with of 16 strips and didn't quite work with the modifications to the fftBin values. We've since added a few similar ones.

ewoudwijma, 2021-03-13:
Thx! Will look to that if I start compiling myself instead of using pre compiled bins (currently having the feb4 version)

closed 2022-05-10 by ewoudwijma