atuline/WLED #70: Apply sound preset at boot

ewoudwijma, 2021-03-12:
Hi Atuline,
If I use a sound effect preset to be started at boot it sometime works and sometimes doesn't.
Is it possible that the Mike isn't initialized at boot procedure?
I have an inmp441

atuline, 2021-03-12:
Not sure what you mean by 'parallel thread'. Also, no mention of which branch or when you downloaded it.

In the dev and master branches (only), I have increased the delay time for startup of 100mS after the drivers are installed to
100mS upon power up and 250mS after the drivers are installed. This was added on March 4th.

ewoudwijma, 2021-03-13:
Forget about parallel thread 😉 (I meant CPU threads, which is not standard supported on arduino, so probably no lt used in WLED).
I am not compiling directly now, i used the bin of feb 4, so running behind the latest version. I hope to find some time in the future to compile latest masters/devs myself.

ewoudwijma, 2021-03-19:
Hi Atuline, I compiled the bin myself using VSC and everything is working like a charm! [image]

closed 2022-05-10 by ewoudwijma