atuline/WLED #71: INMP441 doesn't work when ESP32 DEVKITV1 is powered via VIN

UnchartedBull, 2021-03-14:
**Describe the bug**
Hi, this is a really weird possible bug I've encountered. I have 120 SK6812, so RGB + White Channel, which have been configured and are working as expected. The whole system is powered by a legit Meanwell 10A 5V PSU. +5V and GND from the ESP32 and the LED strips are connected directly to the PSU, which has been configured to output 5.05V. The INMP441 Digital Microphone and LED datapin are wired up according to the wiki.

I've tested the microphone with the provided Sketch and it's definitely working (reading a bit on the lower side - around 500 on normal loudness music). The LED sound-reactive effects are also working if I power everything via USB (removed 60 LEDs for that). If I use the PSU and no USB cable all sound-reactive effects stop working, everything else is working as expected.

I at first thought this might be related to some weird noise in the power system, so I added a cap to the 5V line (35V 1000uF low ESR) which didn't fix the issue. So next I tried connecting the USB cable as well as the Power Supply and all effects including the sound-reactive ones are working. I then unplugged the USB cable and to my surprise everything was still working. Sound-reactive effects are working as expected and I can switch between them, even though the whole system is now powered by the Meanwell PSU. So basically if I "jump-start" the ESP with an USB cable and then switch over to the Meanwell PSU everything works without any issues. After cutting power and restarting the effects won't work anymore. So I ruled any electrical noise issues.

I had a look at the docs but found nothing that could help me. Only interesting thing I possibly found is the initialisation process. My thought here is, that maybe the digital microphone doesn't get initialised fast enough if the ESP is powered via VIN making WLED fallback to analog input (which of course doesn't work).

I haven't tried a custom compiled version yet, my next steps would be to find the mic initialisation in the code and try delaying that by a few seconds to check whether that might solve my issue. Just wanted to run this by you guys before and ask whether you have any ideas / have experienced something similar.

Here is a picture of my wiring if this helps: 
[image]

**Expected behavior**
INMP441 working on VIN power input

**WLED version**
 - Board: ESP32DEVKITV1
 - Version 0.11.1 (latest)
 - Format Binary
 
Thank you for your help!

UnchartedBull, 2021-03-14:
Oh just saw https://github.com/atuline/WLED/commit/9d73163e32be769ababef77e0af68830657e3cf6. Will try building master tomorrow to check whether this fixes my behaviour :)

atuline, 2021-03-15:
Bingo. Hopefully, master or dev should sort you out.

Was also discussed here:  https://github.com/atuline/WLED/issues/70

UnchartedBull, 2021-03-15:
Latest master fixed the issue. Thanks for all your work with the sound-reactive effects, they look really neat :)

closed 2021-03-15 by UnchartedBull