atuline/WLED #72: How to use the new macro system?

JoshWobbles, 2021-03-16:
I would like to have a button act as a momentary switch to a certain preset which will take over until the button is released and then return to previous program.

Use case is I use this on a motorcycle and would like to use your fork to add the sound reactive functionality, but I also like to wire in my brake light to turn all the leds red when braking.

softhack007, 2022-06-18:
It seems that this question is not specific for the "sound reactive" fork of WLED. Did you already ask your question to upstream https://github.com/Aircoookie/WLED/issues?

softhack007, 2022-06-25:
Closing this problem report, it seems to be 'stale'.

You may want to join the discord for getting community  support: https://discord.gg/4CQRmfR

closed 2022-06-25 by softhack007