atuline/WLED #74: Support for magic home dual SPI controller

Nickbert7, 2021-03-23:
Is it possible to add support for the magic home dual spi controller?
It has an integrated microphone and two connectors for LED strips.
https://www.aliexpress.com/i/4001229636389.html
https://www.youtube.com/watch?v=FFR_N8pf03I
[image]
[image]
[image]
[image]

atuline, 2021-03-23:
If it has an ESP8266 or ESP32 and one of the microphones we use, AND if it can be flashed, then that's likely, as you would just need to re-assign the pins to match the unit.

I saw an ESP8265 on the board, so it's highly unlikely. If you can get regular WLED running on this, then our SR WLED has a good chance.

softhack007, 2022-07-14:
this seems to be a stale report - no activity since more than a year.

Currently we can only support ESP32 devices in SR WLED, due to the raw "horse power" needed to run our sound analysis routines.

closed 2022-07-14 by softhack007