atuline/WLED #75: Add log file info to the Info page

MarkinBrisbane, 2021-03-28:
This feature suggestion is quite simply to add the log file info to the info page.

If I boot up Wled when it is connected to my computer, the log file shows info as follows:
[09:35:00]eI2S driver installed.
[09:35:00]Digital microphone is NOT present.
[09:35:00]Ada

My suggestion is to simply add those lines to the Info page, or even a "Show boot log" button to the info page.  If further information is useful then that could also be added.

jlegen, 2021-03-30:
...or perhaps just add an info of the audio hardware found on the info page?

softhack007, 2022-07-14:
A "Show boot log" might be a good idea, however not in scope of our soundreactive branch. 
Also the message 'Digital microphone is NOT present' is not part of the current version any more.
Showing info about "audio hardware found" (plug & play style) also is not possible currently, as it is not clear (technically) how we could properly identify the hardware you use.

closed 2022-07-14 by softhack007