atuline/WLED #76: 2D led configuration not in 0.12.0

ewoudwijma, 2021-04-07:
After installing the new 0.12.0 binary the 2D led with and heights cannot be set in the led configuration settings anymore. As a consequence the 2D effects do not work anymore (all lights blinking red only).

Is it possible they got lost in merging with aircookie code?

bigu1975, 2021-04-08:
confirmed gone for me as well

madmic1314, 2021-04-08:
Confirmed as well. Did an upgrade so previous config are still there. But no settings option. Will be an issue with new configs

merkisoft, 2021-04-13:
i tracked down the commit to this one where it got lost: https://github.com/atuline/WLED/commit/4f367b54010d6b53e60677f496127d6c9e16d01f

try multipin or revert like i did :)

simon-hall, 2021-04-28:
Is this an 'easy' fix for a noob like me?  Or is it better to be patient and wait for the next release?  I'm eager to try out the 2D effects :)

If it's easy, can someone add a bit more detail to explain what to do?

atuline, 2021-04-28:
IIRC, we're just waiting for the guy that worked on the 2D parameters to
integrate his fixes into a release.

In the meantime, the previous version should work just fine.

madmic1314, 2021-04-28:
> Is this an 'easy' fix for a noob like me? Or is it better to be patient and wait for the next release? I'm eager to try out the 2D effects :)
> 
> If it's easy, can someone add a bit more detail to explain what to do?

Try this: flash the old version, erase the settings and then config as you need. Do an OTA upgrade. The settings are missing in the GUI but still working. Worked for me. I needed the newer version for the MIC bug fix that stopped boot.

atuline, 2021-04-28:
No easy fix. You could, however try compiling our dev or even master branches. I'm not sure what their status is right now.

simon-hall, 2021-04-28:
Thanks for the tips - flashing old version and upgrading worked for me.

ewoudwijma, 2021-04-29:
I just played around for a while so just a beginner as well. Recently I managed to compile it myself using MS Code / PlatformIO. I compiled the dev of 20210419 which has the settings in it again. Shared the bin below which you can also install using OTA upgrade. @atuline, is this ok to share intermediate builds like this? Or is there another way in github to deal with intermediate builds?
[ws0.12.0dev20210419.bin.zip](https://github.com/atuline/WLED/files/6398113/ws0.12.0dev20210419.bin.zip)

atuline, 2021-04-29:
Go ahead and share away. THATDONFC who (among other things) makes the builds/releases gets pretty busy at times, while my focus has always been on the FastLED stuff and how to make it work with volume/FFT input.

closed 2022-05-10 by ewoudwijma