atuline/WLED #77: Alexa control

Raito40, 2021-04-07:
Hello,

I'm using your WLED version with a little mic and it's working well, thank you for the work :)
I was trying to add the strip on my Alexa device to control on/off and static colors, but I can't manage to get it detected by my echo dot.
I was wondering if your project work with this Wled functionality.
I'm able to control another original Wled with Alexa, but the audio reactive is not detected no matter what I try.
I'm running it on an ESP8266.
Thanks in advance for any help.

Leon751, 2021-04-14:
I have the same problem. I have activated Alexa and infrared in the code. Then recompiled. Alexa and infrared are not working. the new version is only for the esp32. I have built some lamps with the esp8266 and would like to control them with Alexa. if infrared still works it too would be great. I am using the last version for the esp8266.

hene42, 2021-05-15:
Alexa doesn't work with me either. With the original WLED software, however, it works.

atuline, 2021-05-15:
We've disabled Alexa, among others. You'll need to enable Alexa functionality in wled.h and re-compile from source:

https://github.com/atuline/WLED/blob/d6dad02e305e9a2b5372494b1d8a4b24760c3a77/wled00/wled.h#L30

closed 2021-05-15 by atuline