atuline/WLED #79: Instance (Node) list button missing

Jan2220, 2021-04-10:
**Describe the bug**
The [Nodes] Button from the original v0.12 WLED is missing in this work.
the Audio Reactive ESP32 is also not visible to the default WLED esp8266´s Nodes list. 
Manual OTA Update was successful. 

**To Reproduce**
Update esp32 audio reactive wled to v0.12
Check presence of the [Nodes] Button in the Top Bar.

**Expected behavior**
Nodes Button should be visible and the esp32 should be visible for other esp8266´s?

**WLED version**
 - Board: ESP32 dev
 - Version 0.12.0
 - Format Binary

**Additional context**
//

Thank you for your help!

TarcisioMenezes, 2021-08-28:
Same for me here.! Whenever I enable these settings, my device seems to became really unresponsive. Web UI gets really slow. I had to unable these features, otherwise I couldn't use the Web UI. 

[image]

ewoudwijma, 2022-07-14:
Is this still the case in our latest version (use dev branch). If not , can we close the issue?

closed 2022-07-21 by ewoudwijma