atuline/WLED #80: NO IR-Pin

gitpower2017, 2021-04-10:
Hello.

I cannot configure the IR pin.
It is always set to -1.
I use GIT V0.12 (Sound Reactive)
I have set the following in "platformio.ini" or "platformio_override.in"
I have set the IR function to PIN 4.

 - Board: LEDBoxV2 (esp32)
 - Version: 0.12.0
 - Format: self-compiled

Thank you for your help!

mase76, 2021-04-12:
I have the same issue.

biseyleryapalim, 2021-04-13:
Me too

12 Nis 2021 Pzt 23:02 tarihinde mase76 ***@***.***> şunu
yazdı:

Leon751, 2021-04-15:
Is works with infrared. You must uncomment in the Code
//#define WLED-Disable-Infrared 
at wled.h and then compile.
In Wled sync Interfaces select the infrared remote and aktivate "send notification on button press or ir". 
At last in LED preferences set "IR pin" 4.

Good luck

gitpower2017, 2021-04-30:
After adapting wled.h I can select PIN4 and the remote control (44 buttons) but the IR remote control does not work.
With the latest version of original wled source code, the remote control works.

dirk0034, 2021-05-25:
how do compile??? has somebody a bin-file???

mase76, 2021-05-25:
Here you are.
[esp32dev.zip](https://github.com/atuline/WLED/files/6540802/esp32dev.zip)

closed 2021-05-25 by atuline

dirk0034, 2021-05-25:
thx, but do you have this for 8266?

mase76, 2021-05-25:
Just install VSCodium and PlatformIO. It is really simple. But you have to use the 8266 branch.

atuline, 2021-05-25:
There's also the releases page:

https://github.com/atuline/WLED/releases

mase76, 2021-05-25:
You enabled IR in these builds?

dirk0034, 2021-05-25:
yes, in the 0.12, but after saving it shows -1

mase76, 2021-05-25:
I meant, if atuline enabled the IR pin in the releases?

dirk0034, 2021-05-25:
Oh, I don't know. And by myself I can't compile.
I every time flash the wemos with the bin

dirk0034, 2021-08-20:
Hello again, ist there a .bin Version with IR for wemos d1 mini???

atuline, 2021-08-20:
Have a chat with folks on the Discord channel. Maybe someone has enabled it and compiled it for that.