atuline/WLED #81: ESP32 not enough memory?

Leon751, 2021-04-14:
Hello,
can it be that the ESP32 has too little memory when Alexa or infrared is activated?
I get the error "too little memory" when I activate Alexa or infrared in the Sketch.

Leon751, 2021-04-15:
I have solved the problem. In the arduino ide was the memory setting default. Storage space was too low in the setting.

closed 2021-04-15 by Leon751

atuline, 2021-04-15:
You managed to compile with the Arduino IDE? Good job. I gave up trying to do so several months ago. Your issue would've been a real challenge to find. Congrats on sorting that out.

Leon751, 2021-04-16:
Yes i compile with Arduino IDE 1.8.13. I looked some Videos how to do this. It isnt so difficult. Installed esp32 board and library
Only the memory settings i didnt now.
i use partition scheme  No OTA (2MB APP/ 2MB SPIFFS)
and it worked