atuline/WLED #83: 2D led matrix horizontal or vertical

ewoudwijma, 2021-04-21:
Tried to find how the recommended positioning of a 2D led matrix is. Eg if you do the chase effect, should it go from left to right or from top to bottom? Could not find it in the documentation.

atuline, 2021-04-21:
You know, it's not really been standardized yet. Our 2D flag bearer moved out of country and hasn't been up to programming since. Am hoping someone joins up that can move this forward.

closed 2021-04-21 by atuline

ewoudwijma, 2021-04-21:
Is it okay that we make a decision on it? I want to glue the matrix to a panel and would regret it if later we have animations which are using an orientation. I would propose horizontal then, similar to typing in an editor works (row by row)

atuline, 2021-04-21:
I suspect that when we get someone to work on this, that they'll provide a 2D orientation setting, so you should be OK.

ewoudwijma, 2021-04-21:
Fair enough! For now, on second thought, I will put it with the text (104 in my case) readable, meaning the first led will be bottom left, the last led top right.

ewoudwijma, 2021-04-21:
[image]

emardee, 2021-04-21:
Yes, I think 2D would need a configuration setting, as there are several potential layouts. Zig zag layout of another permutation... i.e. forwards along first row, backwards along second row, forwards along third row... etc. That can make wiring easier, but would need every other "row segment" to be reversed in wled.

"target": "https://github.com/atuline/WLED/issues/83#issuecomment-824075360",
"url": "https://github.com/atuline/WLED/issues/83#issuecomment-824075360",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]

ewoudwijma, 2021-04-22:
True! Just found out that there is a setting already in led preferences for zig-zag called serpentine! I also compiled today the latest dev version and created a copy of an existing 2D effect as a placeholder for me to play around. That worked! So maybe I will contribute in the future 😇. In making it work, I had a problem with expressif32@3.2, that gave an error so changed it back to @3.1.1. Although this is a separate question, is this a known issue (using ms code / platformio)?

atuline, 2021-04-22:
That might be a good question for the main WLED repo folks. We just take their work and add some sound sampling/processing and add a few more animations and controls/settings.

ewoudwijma, 2021-04-22:
Thats true! Will do!