atuline/WLED #85: Binmap effect doesn't work when receiving UDP Sound Sync

exofusion, 2021-04-30:
It seems that the Binmap effect does not work when trying to use UDP Sound Sync to receive microphone data.  The other effects are working fine.  Any ideas what might be going on or anything I could try to fix?

Great project btw!

atuline, 2021-04-30:
Yea, binmap uses the raw 256 unprocessed frequency bins for the animation, and that information is not transferred via UDP sync. We just transfer the 16 aggregated fftResult bins. Just too much data. I've just updated the Wiki docs to reflect that. Thanks!

closed 2021-04-30 by atuline

exofusion, 2021-04-30:
Ah makes sense, thanks for the quick response!

I'm sifting through the code and it looks like the `fftResult` already maps the full set of bins down at 16.  Would it be reasonable to create a new effect `Binmap (Compressed)` to source from `fftResult` instead of the full raw bins, which would then allow a similar effect when receiving remote audio data?  If that seems like a good path, I can try implementing myself and see how far I get

atuline, 2021-04-30:
Feel free to have at it. If you come up with something that works nicely, let us know. You might be able to interpolate values between the fftResult bins or something along those lines. The joy is in hacking out something that surprisingly looks half assed.

In the meantime, I'm looking to see which of the following 2D effects I can add from https://editor.soulmatelights.com/gallery to our dev branch . ..  which eventually gets merged to master by THATDONFC.