atuline/WLED #97: How to get Automatic Gain Control (AGC)

kipz, 2021-05-24:
I'm building a sound-reactive hat. As such, it'll need to work in environments with rapidly changing sound levels without manual intervention.

- Is there a way to enable AGC in sound reactive WLED?
- If not, I'd be happy to attempt a PR, but I could do with some pointers first

I'm generally happy coding/building/testing WLED, but am not sure where the best place in the code would be to integrate such a feature.

Any thoughts or insights most welcome.

atuline, 2021-05-24:
AGC was something I was playing around with, but never really was happy with. Some routines use the AGC I settled on, while others use an Avg or the FFT results. Here's a repository with some of my old AGC work, some of which uses a Proportional Integral (PI) loop:

https://github.com/atuline/FastLED-SoundReactive

In the meantime, if you'd like to do a pull (use our 'dev' branch), which is what I work on, I'd have to test it against all the '*' and possibly '**' routines. Have a look at usermod.cpp and audioreactive.h to see if it's worthwhile.

I'm also up to receive code to review and play around with.

atuline, 2021-05-25:
So, if you do follow up on this, feel free to email me (see my github profile, and I'd be happy to liaise.

terribled, 2021-06-11:
Maybe use a microphone with hardware AGC, like the MAX9814?
https://www.adafruit.com/product/1713

softhack007, 2022-04-11:
You might give the new experimental AGC feature a try -  available in https://github.com/atuline/WLED/tree/dev ; currently supported by only a few effects.
PR https://github.com/atuline/WLED/pull/195
PR https://github.com/atuline/WLED/pull/196

softhack007, 2022-04-14:
I think this issue can be closed in favour of https://github.com/atuline/WLED/issues/170 .

atuline, 2022-04-14:
> Maybe use a microphone with hardware AGC, like the MAX9814? https://www.adafruit.com/product/1713

That won't help with the myriad of other microphones in use, most notably the I2S based ones.

closed 2022-04-14 by atuline