atuline/WLED #100: Enable USERMOD_DHT with SoundReactive WLED

amitav2002, 2021-05-31:
Dear Team,

Firstly, I would like to say Big Thanks for this such amazing and great project. I love it.

All time you guys, helped me out if I needed. Now, Again I am seeking your help in order to enable the DHT22 sensor with my SoundReactive LED.

I am using:
WLED 0.12.0 latest source code
ESP32 Dev Module
DHT22 Sensor

I tried it many times by modifying the source code and complied. But that didn't work for me. I am not sure, where I am doing mistake.

Below is given:-

In usermod_list.cpp, I have defined #define USERMOD_DHT

As documented, if I am copying platformio_override.ini file in root directory, then I am not able to compile the code. Therefore I have modified below in platformio.ino

[env:soundReactive_esp32dev]
board = esp32dev
platform = espressif32@3.2
upload_speed = 921600
;upload_port = /dev/cu.SLAB_USBtoUART
;monitor_port = /dev/cu.SLAB_USBtoUART
build_unflags = ${common.build_unflags}
build_flags = ${common.build_flags_esp32} -D WLED_DISABLE_MQTT -D USERMOD_DHT -D USERMOD_DHT_CELSIUS -D USERMOD_DHT_STATS
lib_ignore =
ESPAsyncTCP
ESPAsyncUDP

And add the library to lib_deps =
https://github.com/alwynallan/DHT_nonblocking

I am able to compile and upload the code, however not seeing AP "WLED-AP".

Please help me out , how I would be able to enable DHT22 sensor with my soundReactive WLED.

You'r response would be highly appreciated.

Thanks very much
Amit Kr. Singh

atuline, 2021-05-31:
Hi, as far as I know, you would need to add your code/library to the current usermod.cpp that we are using.

Keep in mind from the DHT22 page that I just Googled "you can only get new data from it once every 2 seconds". So, it's not like this sensor will have the responsiveness of a microphone, where we're sampling sounds every 100 microseconds and that audio data can change significantly in very short time frames.

amitav2002, 2021-05-31:
Thanks very much for your quick response and valuable advise.

I have defined the library. and complied. Now it looks like working for me.

Thanks again.

closed 2022-07-14 by softhack007