atuline/WLED #103: Relay Wled & Sound

tourer4778, 2021-06-14:
Hello, I use WLed with sound. With Wled without sound you can connect a relay to D6. But that doesn't work with Wled with sound. Is there a way to switch a GPIO when Wled with sound is switched on?

itsboo07, 2021-08-11:
Even I have the same issue... It's not working... Did you find a fix @tourer4778?

softhack007, 2022-07-15:
Hello @tourer4778,

It should work the same way as in WLED without SR. You have to configure the  Relay Pin in Config -> LED preferences. The default value is -1, meaning no relay connected.

[image]

Please try. If that works for you, please close this issue.

closed 2022-07-21 by ewoudwijma