atuline/WLED #104: dev version  LED Preferences Can't save

zx804262156, 2021-06-21:
dev version  LED Preferences Can't save
Save button not working

ewoudwijma, 2021-06-21:
Working on it! Workaround for now: check multiple panels, add 1 horizontal and 1 vertical. Then save should worm

zx804262156, 2021-06-21:
@ewoudwijma 
I find
Button pin
IR pin
Unable to save successfully, pin is -1

atuline, 2021-06-21:
> I find
> Button pin
> IR pin
> Unable to save successfully, pin is -1

The SR version of WLED disables IR. To enable IR, you will need to go into wled.h and 'un' disable it and then compile. Either way, we do not support IR functionality.

zx804262156, 2021-06-21:
> > 我发现
> > Button pin 
> > IR pin
> > 无法成功保存，pin 为 -1
> 
> WLED 的 SR 版本禁用 IR。要启用 IR，您需要进入 wled.h，'un' 禁用它，然后编译。无论哪种方式，我们都不支持 IR 功能。
> 
> 我们还必须确定出现此问题的设备的确切状态，并记录重现问题所需的确切步骤。这不是“保存按钮不起作用”的断断续续的情况，因为它在我们的大部分测试中都在工作。

thank you for your help，To enable IR，will need to go into wled.h,Where to modify，then compile ？I did not understand。

atuline, 2021-06-21:
> thank you for your help，To enable IR，will need to go into wled.h,Where to modify，then compile ？I did not understand。

Summary: We do not support IR and have disabled it. You would need to modify the source code to do so.

ewoudwijma, 2021-06-22:
Solved in latest commit

closed 2021-06-22 by ewoudwijma