atuline/WLED #105: What name is default_envs use appropriate for esp-12 nodemcu v3 8266

tctmaster, 2021-06-25:
i use esp-12e nodemcu v3 8266 so please declare which line is suitable because in .ini only default_envs = soundReactive_esp32dev and eth,
Can I ask more if I use mic max9814, which pin of ESP-32 should I plug in because I have ordered and waiting for delivery, now I only use esp-12 temporarily for testing first?
thanks any bro help me

omski, 2021-09-15:
The support for ESP8266 has been deprecated. I hardly suggest to use an ESP32!
How to hookup analog and digital mics is described in the docs. https://github.com/atuline/WLED/wiki
I would hardly suggest to use a digital microphone!

tctmaster, 2021-09-15:
> The support for ESP8266 has been deprecated. I hardly suggest to use an ESP32!
> How to hookup analog and digital mics is described in the docs. https://github.com/atuline/WLED/wiki
> I would hardly suggest to use a digital microphone!

I'm using ESP32 already and it's solved but I want to ask when will the source code of WLED_Sound 0.12.0 final be available?

omski, 2021-09-15:
https://github.com/atuline/WLED/tree/v0.12.0
Is this what you are looking for?

atuline, 2021-09-15:
We're also waiting for AirCoookie to finish 0.13.0 before making an official release. If the past is anything to go by, we may need to update that as well, as the community finds issues with the official release.

closed 2022-07-14 by softhack007