atuline/WLED #107: Wifi Issues

PassiP, 2021-07-31:
Hi and many thanks for the nixe sound reactive wled project,

i´m very new in wemos stuff and my project starts with some wifi trouble

i use two china nodemcu esp8266 for my wled setups.

after a while using the lights, they are not reachable anymore, after a reboot it works for a while again.
they are showed connected to my wifi but they are not reachable.

any ideas?

best regards

dsyntech, 2021-08-31:
I have the same problem with nodemcu. This is most evident in the version WLED_0.12.0. In 0.11.1, it is more stable.

atuline, 2021-08-31:
We've found the WifFi to be problematic on the ESP8266, and I even re-wrote our SR code from scratch in order to have a minimal footprint. It seems that AP mode is very touchy with the ESP8266 and we never did find out what the issue was.

You might have better response if connecting the ESP8266 to an existing network.

closed 2021-08-31 by atuline

PassiP, 2021-08-31:
I use it already in an existing network and not in AP mode.
Can you recomment a controller with less wifi issues?

datadrian, 2021-08-31:
Esp 32. I've also found a newer router solved many of my esp8266 problems.
I don't know why, but I did.