atuline/WLED #108: How to add TEXT on WLED APP Not use soft 3rd And Help Setting MAP LED on XLights

tctmaster, 2021-08-03:
My board : ESP-32U wroom
Version WLED: 0.12.0 Sound Reactive dev 062221
In 2d matrix on wled of Atuline with settings Serpentine enabled the 2d animation runs to full image no problem but I'm stuck on Xlights
i want to know is there a way to run the TEXT signature right on the WLED UI without going through 3rd software like XLights because i am having trouble in the location MAP LED for XLights because i solder the zigzag serial wire as shown, not straight parallel. and attached picture of my settings for Xlight, can any bro guide me to edit the MAP LED part for Xlight, above on WLED I map as follows: 
{"map":[0, 1, 2, 3, 4, 5 , 6, 7, 8, 9, 10, 11, 12, 13, 14, 15,
16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,
31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,
46,47,48,49,50,51,52,53,54,54,56,57,58,59,60,
61,62,63,64,65,66,67,68,69,70,71,72,73,74,75,
76,77,78,79,80,81,82,83,84,85,86,87,88,89,90,
91,92,93,94,95,96,97,98,99,100,101,102,103,104,105,
106,107,108,109,110,111,112,113,114,115,116,117,118,119,120]}. 
And tried reversing the order of 4 zigzag wires: 
{"map":[0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15,
30,29,28,27,26,25,24,23,22,21,20,19,18,17,16,
31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,
60,59,58,57,56,55,54,53,52,51,50,49,48,47,46,
61,62,63,64,65,66,67,68,69,70,71,72,73,74,75,
90,89,88,87,86,85,84,83,82,81,80,79,78,77,76,
91,92,93,94,95,96,97,98,99,100,101,102,103,104,105,
120,119,118,117,116,115,114,113,112,111,110,109,108,107,106]} ,
but it didn't work correctly, it was overwritten with 2 characters close to each other even though I only showed 1 letter

[image]
[image]
[image]

[image]

>  [image]

ewoudwijma, 2022-07-14:
Seems that no one is picking this up. You still want this or can we close?

gonzogear, 2022-07-21:
I would love this

Drizzt321, 2022-07-21:
At a guess OP means input text directly into the WLED UI for use with Matrixes, rather than have to create them externally and load them?

I'm all for this! Not particularly urgent for me, but I think it'd be a great feature to put on the TODO list, and upstream it.

tctmaster, 2022-07-22:
> At a guess OP means input text directly into the WLED UI for use with Matrixes, rather than have to create them externally and load them?
> 
> I'm all for this! Not particularly urgent for me, but I think it'd be a great feature to put on the TODO list, and upstream it.

I hope Wled will add text playback to WLED for convenience without having to use external soft 3th sync.I currently want to make Led run when going to the stairs and going down, is there any document for me to refer to

Drizzt321, 2022-07-22:
@tctmaster if you mean turn on a preset or playlist when a sensor triggers, you could use the Home-Assistent integration, or use a separate device on your network that could sent a [call to WLED](https://kno.wled.ge/interfaces/json-api/) to tell it to start.

estiens, 2022-08-05:
I would also love an easy way to do scrolling text and seems well implemented in a number of arduino matrix libraries so hopefully shouldn't be too hard to adapt - been a long time since I've written an ino file, but I can take a look over the next couple weeks if no one picks it up - would love to understand more of the internals of WLED anyway

ewoudwijma, 2022-08-05:
Scrolling text is in Blazoncek's fork of WLED

estiens, 2022-08-05:
Wonder if it is easy to merge upstream or relies on too many changes

ewoudwijma, 2022-08-05:
I guess the plan is to move things upstream but what and when is not clear yet

softhack007, 2022-11-28:
Scrolling text is a availeable in [upstream WLED 0.14.0](https://github.com/Aircoookie/WLED), and in [MoonModules  fork](https://AMG.wled.me).

closed 2022-11-28 by softhack007