atuline/WLED #109: ESP8266 branch - When do you merge upstream v0.13.0-b2?

pandel, 2021-08-03:
Hi!

As stated in the subject, I'd like to know, if you will merge upstream to ESP8266 branch in the near future.

Besides, keep up the good work :-) many thanks!

Regards,
Holger

atuline, 2021-08-03:
That will be up to THATDONFC who performs the merges. Have a chat with him on the Discord channels in order to gauge his interest. I think there's been some difficulties with 0.13 so far, and I expect if he DOES merge up to 0.13, that it would be with the official release.

Otherwise, efforts have been focused on the ESP32.

tctmaster, 2021-08-03:
> That will be up to THATDONFC who performs the merges. Have a chat with him on the Discord channels in order to gauge his interest. I think there's been some difficulties with 0.13 so far, and I [...]
hi, can I ask when the full 0.12.0 sound reactive 2d matrix source will be available because source 0.12.0 in github is an old version that is not as complete as the dev version 0622.

pandel, 2021-09-29:
Switched to ESP32 based model.

closed 2021-09-29 by pandel

atuline, 2021-09-29:
> > hi, can I ask when the full 0.12.0 sound reactive 2d matrix source will be available because source 0.12.0 in github is an old version that is not as complete as the dev version 0622.

We're not going to retroactively update 0.12.0 with 2D updates. Version 0.12.0 is what it is. The new 2D code is incorporated into  0.13.X.