atuline/WLED #115: Enabling Audio Sync  WLED confusion flashing

showphi, 2021-09-12:
**Describe the bug**
IF Audio Sync mode selected transmit, WLED confusion flashing, not from the sound, mode disabled WLED is working properly, but cannot be synchronized to other WLEDs.

**WLED version**
 - ESP32
 - soundReactive_WLED_0.12.0a_ESP32.bin
 - MAX4466

Thank you for your help!

softhack007, 2022-07-14:
MAX4466 confusing microphone is, maybe try I2S digital one. Will be better quality. Also please try latest `dev` branch.

Possibly related also to issue #177.

softhack007, 2022-07-20:
Closing this report due to inactivity, and too little information to reproduce the reported problem.

Audio sync was recently improved, please try our [latest `dev` branch](https://github.com/atuline/WLED/tree/dev).