atuline/WLED #116: digital Microphone enabled even if set to disabled in setup screen

omski, 2021-09-14:
**Describe the bug**
I was playing around with line in and digital microphone. I switched off the digital microphone (INMP441) through the settings screen but still got reactions from it. I have to remove the mic from the breadboard to get rid of its input.

**To Reproduce**
hook up a digital microphone and hook up line in to analog input (VP - PIN36)

**Expected behavior**
disabling the digital microphone should disable its input and enable analog input only.

**WLED version**
 - Board: ESP32 Dev Kit v4
 - Version: soundReactive_WLED_0.12.0_dev_0622_ESP32.bin as well as soundReactive_WLED_0.12.0a_ESP32.bin 
 - Format. Binary
 
Thank you for your help!

r312645, 2021-09-24:
Same problem here.

FHeilmann, 2021-11-03:
Hi, I've[ just submitted a PR](https://github.com/atuline/WLED/pull/130) that might resolve the issue you're having. Please check and let me know if the code can be improved further!

softhack007, 2022-07-08:
No additional feedback since several month. Assuming the problem is solved now.

closed 2022-07-08 by softhack007