atuline/WLED #117: ESP32 A1S support

omski, 2021-09-14:
**Is your feature request related to a problem? Please describe.**
Currently WLED SR seem to work only reliably with a digital MEMS microphone. The analog input of the ESP8266 and ESP32 seem to really useful. At least it is very noisy...

**Describe the solution you'd like**
Support for a second digital line input from an I2S ADC.
Like this one https://www.aliexpress.com/item/1005002673407636.html

**Describe alternatives you've considered**
I used the ADC Pin of ESP32 and ESP8266 and even with lots of DC filtering there are spikes and noise. A digital microphone (INMP441) creates decent results but it will react to ambient noise too. A nice and clean line input would be just great i think.

FHeilmann, 2021-11-04:
I had a short look and this will unfortunately be difficult at best and impossible at worst. The reason is that WLED and therefore WLED-SR builds on a version of the `arduino-espressif32` framework where no master clock support exists (this was only added later). 

Updating the `arduino-espressif32` framework, I assume, is out of the question, since that will most likely affect a million other things and has a high chance of breaking everything.

Generating the clock manually might be possible, but that's going to be a larger undertaking because the master clock depends on the sampling frequency, the word select frequency and probably some other things, and then needs to be generated with minimal jitter and without disturbing the remainder of WLED.

So while it's certainly not impossible, I wouldn't hold my breath at this time.

Cheers,

Florian

atuline, 2021-11-04:
Thanks much for that research Florian.

closed 2021-11-04 by atuline

tony-fav, 2021-11-04:
I just created a PR for supporting the ES7243 mic which requires master clock. https://github.com/atuline/WLED/pull/132

It is possible in the current framework to generate a master clock, but it is not as directly readable as the latest ESP-IDF. (Still only 2 lines).

atuline, 2021-11-04:
PR has been merged. Thanks tony and florian for your research, discussion and pulls.