atuline/WLED #118: ESP A1S support

omski, 2021-09-14:
**Is your feature request related to a problem? Please describe.**
My request is hardly related to the poor ADC performance of the ESP. There is another request regarding a second I2S input.
But there is an ESP variant which seems to match perfectly with WLED SR as this project is all about making LED dance to some noise with the help of an ESP.
The ESP A1S is has some special audio capabilities like:
microphone input
line input
audio coded with several supported formats
There is an dev kit available: https://github.com/Ai-Thinker-Open/ESP32-A1S-AudioKit

**Describe the solution you'd like**
Support for the ESP32 A1S and its audio input capabilities.

omski, 2021-09-16:
I've got an answer to my request on espressif/arduino-esp32 in regards to support for the ESP A1S. https://github.com/espressif/arduino-esp32/issues/5674
"if the manufacturer or someone from the community wants to add it, they are free to do so. Otherwise you can use the "ESP32 Dev Board" instead, which is compatible with all other boards."

So as the dev kits for the ESP A1S are reasonable priced one could think of making a version of WLED SR these where the added features of the chip are supported out of the box?
https://www.aliexpress.com/item/32957553086.html
This will make life much easier because the board has already microphones and a line in with an ADC. No more struggles getting good audio signals into the ESP to calculate LED effects on :wink:
Maybe the added features could be made accessible through user mods.

atuline, 2021-09-16:
If the microphone is already there, I would suspect that the pin select feature in WLED SR would allow you to use those features.

omski, 2021-09-16:
There are two of them and I suspect that they are not connected via the analog pin of the ESP instead through MIC input of the unbuild AC101 audio codec chip itself.
[image]
https://html.alldatasheet.net/html-pdf/1134488/XPOWER/AC101/249/1/AC101.html
[image]

Furthermore there is line input going to the AC101 too.

When we can figure out how to access these inputs it would be nice to have a default_env for that device don't you think?

scruffynerf, 2021-09-28:
Supporting the esp32 LyraT (aka the es8388 chip)
As well would be nice. It's i2C

Separate add-on board:
https://pcbartists.com/products/es8388-module/esp32-es8388-audio-codec-interfacing/

omski, 2021-09-29:
@scruffynerf 
Maybe this issue here https://github.com/atuline/WLED/issues/117
suits better to your proposal. It is about support for external I2S/I2C boards for line input.

Nice little add-on board! I am about to use one of these for a test: https://de.aliexpress.com/item/4000983028754.html

tony-fav, 2021-11-04:
This mic should be easily supported. I located this project which seems to use it with config not much different than that currently https://github.com/RealCorebb/ESP32-A1s-Audio-Kit/blob/master/Microphone.ino However, it seems to require an I2C setup step which can be found here https://github.com/Yveaux/AC101/blob/master/src/AC101.cpp

I have just created a PR for the ES7243 which similarly requires an I2C configuration step: https://github.com/atuline/WLED/pull/132

omski, 2021-11-08:
> @omski do you have the board in hand?

Yes it just arrived. I will try your solution as soon as I have some time to switch to the related project back again. Have you any advice for me to assure that I am not messing up with your proposal?
Would be great to have support for this board because it is cheap compared to what it has available. The I2S/I2C ADC add-on boards alone are more expensive.

microfx, 2022-02-17:
did it work after all?

ewoudwijma, 2022-07-15:
Loses due to inactivity

microfx, 2022-07-20:
Yes, please!

softhack007, 2022-07-20:
I think that there is still (a lot) hope for these boards with built-in audio peripherals, like

* Ai-Thinker ESP32 Audio Kit https://docs.ai-thinker.com/en/esp32-audio-kit
* ESP32-LyraT V4.3 https://docs.espressif.com/projects/esp-adf/en/latest/design-guide/dev-boards/get-started-esp32-lyrat.html

netmindz, 2022-11-27:
I've made a start on the ES8388 support https://github.com/netmindz/WLED/tree/ES8388

netmindz, 2022-11-27:
https://github.com/MoonModules/WLED/pull/5

netmindz, 2023-04-16:
This is now supported in MoonModules, do you need it back-porting to SR @omski ?