atuline/WLED #119: sync multiple instances with realtime sync or DMX

omski, 2021-09-14:
**Is your feature request related to a problem? Please describe.**
Sync multiple instances as 1:1 copies of the master instance is currently only possible though audio sync which has some limitations.
WLED SR has the audio sync feature but the receiving instances must be WLED SR instances and there is a feature gap between ESP8266 and ESP32 devices.
Currently WLED and WLED SR is not able to sync multiple instances via realtime UDP or DMX.

**Describe the solution you'd like**
The master instance should create realtlime UDP or DMX data and send it to every slave instance. This way you can use 'plain' WLED slave instances to receive effects created by WLED SR.
Sync via sending audio to any slave is fine for creating different effects on the same audio data but this is not needed if someone just wants to copy the effect of an instance to another instance.

atuline, 2021-09-15:
I believe that in order to do this, we would need to control the slave devices much like LedFX controls remote devices and that is to send all of the LED information to the remote device. That's not currently on our roadmap, however we'd be open to a comprehensively tested PR that provided this functionality along with the GUI selectability.

closed 2021-09-15 by atuline

netmindz, 2021-09-18:
In effect you mean the option to send e1.31 or one of the other "raw" LED value sync options.

While that's not SR specific, I think you would find hard to get accepted upstream.

For the SR branch the usecase will be restricted to cases where all your devices have the same LED count as using the normal pattern sync + audio data is more flexible than your proposal.

Depending on your physical setup, simply connecting multiple strips to the same controller might be more appropriate, using max485 if you have longer cable lengths

omski, 2021-09-18:
It is SR specific as makes only sense with WLED SR because you need audio data to render the effects. WLED has only synthetic effects which are easy to reproduce on different instances. With WLED SR your furthermore limited du to version compatibilty issue in regards to the firmware and the chip used.

My physical setup does not allow for wired connections because the receiving targets are moving. They will be battery driven. For now the limitation of having identical setups is not a problem for me.
But I would propose to split of segments of the master device to several slaves like WLED does this with sync groups.

I might be helpful to imaginate a master instance wich is directly connected to a music source through a line in signal and several moving slave device which are not able to source the audio signal themselves and are space restricted, so an ESP32 chip would be hard to cram into. This is the reason I have to use ESP8266 WROOM 2U as they are have very small dimensions.