atuline/WLED #121: Consider using Framebuffer::GFX for 2D support

marcmerlin, 2021-09-16:
I looked at WLED in the past, but didn't use it because I use 2D and it was 1D only.
I see that you are adding 2D support now (cool).

Could you have a look at this?
* http://marc.merlins.org/perso/arduino/post_2020-03-16_Framebuffer_GFX_-Choosing-between-its-3-2D-APIs_-FastLED-XY_-NeoMatrix_-and-LEDMatrix_-and-detail-of-its-many-supported-hardware-backends.html
* https://github.com/marcmerlin/Framebuffer_GFX

This will give you 2D support on multiple hardware backends (TFT, RGBPanel, FastLED, etc...)
It will also give you 3 APIs, FastLED, LEDMatrix, and Adafruit::GFX

netmindz, 2022-02-24:
Just trying to support patterns designed for FastLED is tricky due to the way the upstream WLED is written, trying to maintain a fork where we have to port to your API would be unmanageable. 

Feel free to suggest upstream as would be better placed there

embedded-creations, 2022-02-24:
I'm going to take a second look at this as the Mapping class I'm working on (for upstream WLED) will make adding a Virtual Screen compatible with Marc's Framebuffer class a lot easier.  No need to keep this issue open for my sake though.

marcmerlin, 2022-03-14:
Thanks @embedded-creations . If you need any help making a framebuffer::gfx subclass, let me know, but it's reasonably trivial and I'm pretty sure your C++ skills are superior to mine anyway :)

closed 2022-07-21 by ewoudwijma