atuline/WLED #125: Matrix multiply must match exactly

scruffynerf, 2021-09-30:
**Describe the bug**
Bug is confirmed in latest 13b3.  If you put it a matrix value (width x height) less than the total # of pixels, it stops working correctly and only does remainder as matrix.  So I have 300 pixel strip, in lamp spiral.  If I enter 11x27 (which is 297, closest value to 300), it only lights up 3 columns of lights.
Changing total pixels to 297, works as expected, rest of leds light up.

**WLED version**
 - Board: Quindor DigUno w ESP32
 - Version 13b3
 - Format - released binary

dosipod, 2021-12-22:
I am trying to simulate this on esp32 with 13b4 using 8x32 matrix  and unfortunately any **major** changes to width and height results in occasional led strange behavior and sometimes esp crash . I thought this is related to physical matrices  and HW setup .

softhack007, 2022-09-02:
Is the bad behaviour still observable when you use our latest release 0.13.2 ??
https://github.com/atuline/WLED/releases/tag/v0.13.2

softhack007, 2022-09-07:
Closed as no activity in this issue.

closed 2022-09-07 by softhack007