atuline/WLED #131: Failed updating firmware!  Please check your file and retry!

glowingkitty, 2021-11-03:
I compiled the latest /dev version of your WLED fork and uploaded it to my ESP32 running your previous WLED fork (/dev) from a few days ago - by using the OTA update feature of WLED. But it always crashes... any idea why?

"Failed updating firmware!
Please check your file and retry!"

atuline, 2021-11-03:
Afraid not. Feel free to join the Discord forums to discuss further. According to one of the developers, there WAS an issue a while back, but it should be OK now. That being said, that same developer just added an interpreter, which needs a LOT of work.

In the meantime, when updating, you should always perform a 'Security & Updates | Factory Reset'.

ewoudwijma, 2021-11-04:
'One of the developers' might have an idea why this is ;-)
Recent versions have the following line in their platformio while older versions didn't have that: 
board_build.partitions = tools/WLED_ESP32_4MB_1MB_FS.csv
So can you try if it works if you comment that out?

Angel-1024, 2021-12-08:
> 'One of the developers' might have an idea why this is ;-) Recent versions have the following line in their platformio while older versions didn't have that: board_build.partitions = tools/WLED_ESP32_4MB_1MB_FS.csv So can you try if it works if you comment that out?

commenting it will cause compiling error
"Error: The program size (1329890 bytes) is greater than maximum allowed (1310720 bytes)"

Compile with this option and flash firmware via cable, not by OTA, this way works.

atuline, 2021-12-08:
When I compile, I now use env:soundReactive_esp32dev when compiling, again, not OTA.

closed 2021-12-08 by atuline