atuline/WLED #135: Hi there! I´m unable to make work microphone in the m5stickc,everithing works as expected except that point.anyone tryed this before?

hediondo1, 2021-11-11:
**Take a look at the wiki and FAQ, perhaps your question is already answered!**
[FAQ](https://github.com/Aircoookie/WLED/wiki/FAQ)

**Please consider asking your question on the WLED forum or Discord**
[Forum](https://wled.discourse.group/)
[Discord](https://discord.gg/KuqP7NE)
[What to post where?](https://github.com/Aircoookie/WLED/issues/658)

**If you do not like to use these platforms, delete this template and ask away!**
Please keep in mind though that the issue section is generally not the preferred place for general questions.

hediondo1, 2021-11-11:
anyone tryed to make work m5stickc mic? i´m unable to make it work. i achieved m5 atom matrix with both matrix and external led simultaneous and external analog mic but m5stickc not working with mic.

FHeilmann, 2021-11-15:
Unfortunately the mic on the m5stickc is a PDM i2s microphone, which is currently not supported by WLED-SR

FHeilmann, 2021-12-31:
@hediondo1 please do give the latest `dev` branch a try. I added pdm i2s microphone support 1-2 weeks ago.

hediondo1, 2022-01-02:
ok i´ll try soon,thanks for your efforts!! good work!!

hediondo1, 2022-01-06:
ok atom echo,atom matrix with embeded matrix +some leds working with digital ext mic.m5stick c working but need to check the screen now.
atom echo works with pdm mic perfect.looks a good choice for soundreactive wled.
next step make stickc full working.

claudiosv, 2022-05-17:
I'm working on getting the M5Stick C to work again in #206 .

softhack007, 2022-06-18:
Hi @hediondo1, did the latest DEV version (with PDM microphone support) work for you? If yes, can we close this issue?

softhack007, 2022-06-28:
WLED dev version supports PDM microphones, and the problem report author confirmed it works for him. Assuming that the problem is solved now.

closed 2022-06-28 by softhack007