atuline/WLED #141: Loop crash on boot with M5Stamp Pico

domisjustanumber, 2021-11-23:
I'm excited to try out the SR version of WLED after using the regular version for quite a while, but discovered an issue when trying to use the cheap M5Stamp Pico boards.

Not sure if this is a bug or feature request, as it's not a listed piece of supported hardware, but as the regular version of WLED works, and I figured I'd put it as a bug for now, but feel free to move it to Feature Request.

**The bug**
I flashed WLED Sound Reactive onto an M5Stamp Pico https://docs.m5stack.com/en/core/stamp_pico and it goes into a loop crash on boot almost immediately.

The regular non-sound reactive version of WLED (at least 0.12.0) works as expected.

**WLED version**
I have tried the following versions of WLED SR with the same results.
0.12.0
0.13.0-b1
0.13.0-b4

I tried flashing 0.13.0-b1 via https://install.wled.me/
and the other versions using ESPHome Flasher

**Log**
[18:00:18]Rebooting...
[18:00:18]ets Jun  8 2016 00:22:57
[18:00:18]
[18:00:18]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[18:00:18]configsip: 188777542, SPIWP:0xee
[18:00:18]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[18:00:18]mode:DOUT, clock div:2
[18:00:18]load:0x3fff0018,len:4
[18:00:18]load:0x3fff001c,len:1044
[18:00:18]load:0x40078000,len:8896
[18:00:18]load:0x40080400,len:5828
[18:00:18]entry 0x400806ac
[18:00:19]Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
[18:00:19]Memory dump at 0x4018ebfc: 22390020 6580ffec 6910fc0c
[18:00:19]Core 1 register dump:
[18:00:19]PC      : 0x4018ec02  PS      : 0x00060730  A0      : 0x801896b5  A1      : 0x3ffb1db0  
[18:00:19]A2      : 0x3ff440c0  A3      : 0x00000000  A4      : 0x00060b20  A5      : 0x00000001  
[18:00:19]A6      : 0x00060b20  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3ffb1d80  
[18:00:19]A10     : 0x00000000  A11     : 0x00060b23  A12     : 0x00060b20  A13     : 0x00000001  
[18:00:19]A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000010  EXCCAUSE: 0x00000000  
[18:00:19]EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  
[18:00:19]
[18:00:19]ELF file SHA256: 0000000000000000
[18:00:19]
[18:00:19]Backtrace: 0x4018ec02:0x3ffb1db0 0x401896b2:0x3ffb1de0 0x40189b15:0x3ffb1e10 0x400e5dfd:0x3ffb1e50 0x400e5f7a:0x3ffb1ea0 0x400e8513:0x3ffb1ed0 0x400e87e2:0x3ffb1ef0 0x400fc731:0x3ffb1f20 0x400fc841:0x3ffb1f40 0x400fe15a:0x3ffb1f90 0x4011cd8a:0x3ffb1fb0 0x4008acca:0x3ffb1fd0
[18:00:19]
[18:00:19]Rebooting...

atuline, 2021-11-23:
I believe one of the submitters was looking into alternate inputs i.e. from this board. No timeframe though, because when you patch one issue, 5 more crop up.

domisjustanumber, 2021-11-24:
Thanks for the quick reply Andrew!

I did some more tinkering with this and I got it working. It seems to be an issue in the flashing process and not the code itself, so as I think this is a generic WLED issue rather than a sound-reactive specific one, so I've put my findings here: https://github.com/Aircoookie/WLED/issues/2035

atuline, 2021-11-24:
> Thanks for the quick reply Andrew!
> 
> I did some more tinkering with this and I got it working. It seems to be an issue in the flashing process and not the code itself, so as I think this is a generic WLED issue rather than a sound-reactive specific one, so I've put my findings here: [Aircoookie#2035](https://github.com/Aircoookie/WLED/issues/2035)

What a great write-up. Thanks. Interestingly, I was looking for a way to manually flash devices, because our sound reactive version of WLED is getting into different partition sizes and the ESP Home Flasher I'd been using for the ESP32 binary is not configurable. Your write-up may be just the ticket!!

closed 2021-11-24 by atuline