atuline/WLED #147: Matrix and single strip won't work together.

Funland-James, 2021-12-06:
I have a lamp that has a matrix on the bottom and a 28 LED strip on the top. It uses an ESP32 and has a separate output for the bottom section (pin gpio5) and the top section (gpio16).

When running under standard WLED both outputs work as individual strips. Under SR Wled, when I define the matrix and then define the single strip, only the matrix works. The top 28 LEDs do not light at all.

To reproduce, connect a matrix to a data pin on the esp32 and connect a strip to a different data pin. Define accordingly under "LED Preferences". Only the Matrix works. The strip never lights up.

The expected behavior in that both the Matrix and the strip are controllable separately under Segments.

**WLED version
 - Board: Esp32 mini NodeMCU
 - Version: 0.13.0-b4
 - Format: Binary

I initially did an upgrade. When it didn't work I did a complete reinstall of SR Wled .13 - b4 and got the same behavior.

Thank you for your help!

ewoudwijma, 2021-12-07:
It is not possible to have both a matrix and a strip, when you define a matrix all pixels are used for the matrix where ledCount = matrixWidth x matrixHeight

Funland-James, 2021-12-08:
If you can't have both then it should be reflected in the interface. As it is now, you are able to define both at the same time. It just doesn't work. I don't know why this would be a problem if you use different gpio pins for each.

al3x1337, 2022-03-11:
I second this. I'm building a 3D Printed Clock Shelf and plan to use 3 outputs: One for the horizontal runs, one for vertical runs and one for downlights. 
If there was a way to use segments for a matrix, I could define my horizontal runs as a 3x6 matrix and use only those.

ewoudwijma, 2022-07-14:
Nobody picket this up. Do you want to keep it open? Or can we close? (Only thing what is planned for later this year is that each (same nr of leds) panel can be configured separately one of the panel can be non serpentine and used for a strip)

Funland-James, 2022-07-15:
I suppose it can be closed. I have had to simplify my designs because this feature is missing and I really wish it was supported. I'm surprised that nobody else has reported it, but if I'm the only one then kill it.

ewoudwijma, 2022-07-15:
Closed as no code change planned

closed 2022-07-15 by ewoudwijma