atuline/WLED #148: LED Preference page

Larsen23, 2021-12-14:
Good morning
I have a problem with the LED Preference page. If I create 7 outputs everything is normal. If I create 8 outputs the page is blank. This is only in the SR version, I do the same with the "normal WLED version", the error is not

[image]

[image]

.

jonandel, 2021-12-21:
I get that too - both slow loading of that page, or sometimes blank (my guess is a timeout) - depending upon what preset/effects etc is being shown - and it seems to be sensitive to WIFI strength too.... methinks the 8366 (in my case) is struggling...

PolkaPolk, 2022-01-01:
I have the same problem. The ESP32 only shows a blank page.

atuline, 2022-01-01:
Don't forget that:

- The ESP8266 has a separate branch
- If you're flashing a binary, we are no longer using the default paritioning. See the updated installation page.
- If you've done any upgrades, you should perform a factory reset.
- See the troubleshooting page https://github.com/atuline/WLED/wiki/My--device--is-not-working

Larsen23, 2022-01-02:
Hi
I wish you a healthy new year. 
I use an ESP32. I've tried install.wled.me and the problem is the same. Whenever I enter all data, the page is no longer indicated. If I then install the normal WLED version, I get straight back to the page without any problems.

atuline, 2022-01-02:
> I use an ESP32. I've tried install.wled.me and the problem is the same.
You may not be able to use install.wled.me. Again, please see our installation page.

PolkaPolk, 2022-01-03:
Hi
I've tried to install it with the instructions on your installation page but the error is still the same. The normal WLED version is still working.

Larsen23, 2022-01-03:
Hi

Yes that's right. it is always the same with me. I finally gave up. If I want to change something now, I install the original, make the change and reinstall the sound version. Unfortunately I haven't found another way.

atuline, 2022-01-03:
OK, just confirming that you both got an ESP32 binary and ran:

esptool.exe write_flash 0x0 esp32_bootloader_v4.bin

esptool.exe write_flash 0x010000 soundReactive_WLED_0.13.0-b4_ESP32.bin

and were unsuccessful.

Larsen23, 2022-01-03:
Yes that's right. Today after work I can also make a video of the process so you can see all the steps.

atuline, 2022-01-03:
OK, thanks. I'll review with the folks on our Discord beta test and dev channels.

Larsen23, 2022-01-03:
here is the video.

[https://drive.google.com/file/d/1kh-y9KGzJgeRJlGR01HqU4fcevRBrHUV/view?usp=sharing](url)

THATDONFC, 2022-01-04:
I can confirm this bug existed in SR-WLED-b4. It does not exist in SR-WLED-b6. Please flash the most current version of SR-WLED-beta and report back.

Larsen23, 2022-01-04:
Hi
I can say that with version b6 I come to the Settig's side. Tomorrow I will test it on the ESP32 which is also installed with the LED. But at first it looks so right. 

thank you

softhack007, 2022-07-14:
fixed sinceSR-WLED-b4.

closed 2022-07-14 by softhack007