atuline/WLED #152: Windows Client

FreshImmuc, 2021-12-20:
Ok so Its just a simple feature. Just add a Windows client that connects to a WLED Chip and uses the internal audio from windows as the audio input. Its just that this would be so great for everyone who just listens to music on his pc and not loud with music boxes.

atuline, 2021-12-20:
LedFx handles that use case.

This project is more for a portable device with direct analog or I2S input. 

That being said, if you'd like to write that software for Windows, our UDP sound sync documentation is at https://github.com/atuline/WLED/wiki/UDP-Sound-Sync.

closed 2021-12-20 by atuline

FreshImmuc, 2021-12-20:
Yes, its a great project here but not very useful for me since I need Desktop Audio. I tried this rn and it works so good job. :)