atuline/WLED #154: 0.13.0-b4 unrecoverable crash in Clock Overlay

dosipod, 2021-12-22:
Hi

I have tested multiple esp32 ( Nodemcu and DEVKITV1) running wled 0.13.0-b4 .First thing I do is a fresh install of wled 0.13.0-b4 , I only change wifi setting to connect to my home network then :
Go under Config -->Time & Macros -->Clock -->Clock Overlay -->Analog Clock ( then save )

MCU will go into unrecoverable crash , rebooting or powering down MCU does not work and it will loop rebooting with the below message ( viewed via usb esphome-flasher ) . The only solution to recover the MCU was to re-flash it again . I tried this 3 time and result is the same

Can someone test on the same version and confirm ? please test only if you are willing to re-flash the MCU .Because the environment i tested with is not production i just wanted to confirm if this is a known issue . If this is a known issue then please just send link

There is an error for cpu panic :
Guru Meditation Error: Core 1 panic’ed (IntegerDivideByZero). Exception was unhandled.
below is repeated so i only attached one page of it

[11:01:56]Rebooting…
[11:01:56]ets Jun 8 2016 00:22:57
[11:01:56]
[11:01:56]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[11:01:56]configsip: 0, SPIWP:0xee
[11:01:56]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[11:01:56]mode:DOUT, clock div:2
[11:01:56]load:0x3fff0018,len:4
[11:01:56]load:0x3fff001c,len:1044
[11:01:56]load:0x40078000,len:8896
[11:01:56]load:0x40080400,len:5828
[11:01:56]entry 0x400806ac
[11:01:57]I2S driver installed.
[11:01:57]Digital microphone is NOT present.
[11:01:57]Ada
[11:01:57]Guru Meditation Error: Core 1 panic’ed (IntegerDivideByZero). Exception was unhandled.
[11:01:57]Core 1 register dump:
[11:01:57]PC : 0x400dc6cb PS : 0x00060d30 A0 : 0x800f4875 A1 : 0x3ffb1e20
[11:01:57]A2 : 0x3ffcb808 A3 : 0x0000000e A4 : 0x00000000 A5 : 0x0000000e
[11:01:57]A6 : 0x3ffcb808 A7 : 0x7ffc0000 A8 : 0x00000000 A9 : 0x000006ba
[11:01:57]A10 : 0x00000401 A11 : 0x00000000 A12 : 0x00000000 A13 : 0x3ff00000
[11:01:57]A14 : 0x00000000 A15 : 0x00000000 SAR : 0x0000001c EXCCAUSE: 0x00000006
[11:01:57]EXCVADDR: 0x00000000 LBEG : 0x400f9ec1 LEND : 0x400f9ee1 LCOUNT : 0x0000000b
[11:01:57]
[11:01:57]ELF file SHA256: 0000000000000000
[11:01:57]
[11:01:57]Backtrace: 0x400dc6cb:0x3ffb1e20 0x400f4872:0x3ffb1e40 0x400f4c6b:0x3ffb1e70 0x400f4cdf:0x3ffb1ec0 0x400e4633:0x3ffb1ee0 0x400e5dc5:0x3ffb1f00 0x400fc99e:0x3ffb1f40 0x400fe15a:0x3ffb1f90 0x4011cd8a:0x3ffb1fb0 0x4008acca:0x3ffb1fd0
[11:01:57]
[11:01:57]Rebooting…

dosipod, 2021-12-23:
I got a reply from  https://www.reddit.com/user/johnny5canuck/ 
That ESPHome-Flasher could be the issue with 0.13.0-b4 and we should use ESP32 bootloader as mentioned below
https://github.com/atuline/WLED/wiki/Installing-and-Compiling#flashing-esp32-binaries

dbizarro, 2022-01-03:
Hi, same behavior.

Flashed wled 0.12 through the Web installer. Ota update to sr 0.13.b4. every think works ok until i set up clock overlay.

dosipod, 2022-01-04:
Hi ,I think that  confirms the issue on b4 but I stopped testing the same because the crash is unrecoverable in my case  and the issue is fixed in b6 so i guess it is  a known issue and hopefully fixed in the next release . 
Cheers

dosipod, 2022-02-01:
Closing this as the issue has been fixed in b6

closed 2022-02-01 by dosipod