atuline/WLED #155: v0.12.0 ESP32 800 LED's :: Creating Segment

gniknalu, 2021-12-24:
**Describe the bug**
When creating segments, it won't "keep" them after I close or when I inactivate one or leave the menu and return, the "0" segments reverts to defaults.

**To Reproduce**
On a ESP32 I've connected 3 WS821x strips (GPIO16, 18 and 19) with a total of 800 LED's. Trying to create segments. 
Segment 0: 0 - 264
Segment 1: 265 - 536
Segment 2: 537 - 800

After defining them, and I click on another menu (like color or effects) or if I inactivate one segment, the 0 segment reverts to Start LED 0/Stop LED 800

This happens on the Android app or when accessing it via http in my browser

**Expected behavior**
After defining a segment, the segment information should be retained when you leave the menu, disable a segment or adjust the segment (brightness or "reverse direction or "mirror effect")

**WLED version**
 - Board:  ESP32 (ESP-WROOM-32)
 - Version: v0.12.0 "Hikari"
 - Format: Binary
 - Microphone: LM386 Electret Microphone (*yes it works!*)

I appreciate this fork and thank you for your assistance!

atuline, 2021-12-24:
Whenver I have an issue, I'll make sure I've done a 'Factory reset' to ensure the device is absolutely fresh.

In addition, I advise testing with the same version of stock WLED, with that Factory reset performed as well.

gniknalu, 2021-12-24:
> Whenver I have an issue, I'll make sure I've done a 'Factory reset' to ensure the device is absolutely fresh.
> 
> In addition, I advise testing with the same version of stock WLED, with that Factory reset performed as well.

I am on my 3rd ESP32 and had the same experience on the other 2. My "hack" is to define the first segment (Segment 0). Click out to something else (Effects or Color). Return and define the second etc.   Shouldn't have to do this - hence why I opened this report.

atuline, 2021-12-24:
The question about 'Factory reset' still remains, hence why I recommended it. Shouldn't have to repeat myself.

atuline, 2021-12-24:
I was able to replicate this issue on our dev SR branch by:

- Compile/upload fresh version.
- Perform a factory reset.
- Configure for 800 LED's.
- Add the segments.

I then did the same with 0.13.0 b5 from AC's master and had the same results. Our fork is taken from the Master at wled.me and when he fixes that, we'll incorporate his updates (pending availability of @THATDONFC who performs the merges).

I recommend raising the issue at wled.me.

closed 2021-12-24 by gniknalu

atuline, 2021-12-30:
By the way, THATDONFC just merged a 0.13.0b6 with our SR fork and that Segment issue goes away. It's currently at:

https://github.com/atuline/WLED/tree/dev-b6-merge

and he's provided binaries in our beta testing Discord channel.