atuline/WLED #156: Works on DEBO JT nodeMCU ESP32 but reboots when acccessing WEB-UI

DarthWeber, 2021-12-27:
i have an issue with a DEBO JT nodeMCU ESP32 flashed with Sound Reactive v013-b4. I connected 30 LEDs long strip WS2812B to GPIO13. No micro added until now. No usermods.

If i open the web-ui the nodeMCU seems to interrupt or even reboot. The web interface drops the red “connection to light failed” - but after a short time everything runs as expected.

While monitoring the connection with ping i can observe that the ping connection is hold for a few seconds.

This happens always on the first access to the Web-UI

Any ideas what could be wrong ?

atuline, 2021-12-27:
How did you flash it? See:

https://github.com/atuline/WLED/wiki/Installing-and-Compiling#flashing-esp32-binaries

ESPHomeFlasher works with the original WLED, but no longer with SR WLED because we've grown.

DarthWeber, 2022-01-07:
Thanks. Correct flashing solves this.

closed 2022-01-07 by DarthWeber