atuline/WLED #158: line-in wiring to pre-out

DarthWeber, 2022-01-03:
HI, unfortuanly i have no monitor or line-out signal at my denon amplifier. But there is a pre-out which i would like to use as line-in. But if i wiring it like the wiki, there is no reaction to the signal. Probably the signals on pre-out are too low (should be less than 1V)

Does anybody know how i could use the pre-out signal of my amplifier ?

Thanks,
Tom

atuline, 2022-01-03:
You might also want to join up on the Discord to discuss.

dosipod, 2022-01-22:
I have tested 3 setups with line-in ( direct line-in to GPIO39 or GPIO36)  from 3.5mm jack , so nothing like the wiki and no resistors or capacitor and all the 3 setups worked ( the gain had to be set to around 90 to 150   )  . Is the wiki line in wiring a recommendation like the level shifter or mandatory ? I have tried to look on Espressif  forum for esp32 line-in wiring but with no luck

atuline, 2022-01-22:
If it works, good for you. For the provided circuit, the 680 ohm resistors provide terrmination so that you don't get reflection on the signals. The 100nf capacitors remove any DC offset from the incoming signal. Remember that the ESP goes from 0V to 3.3V. Beyond that are dragons. Finally, the 1M resistors provide DC centering of the incoming (now) AC only signal halfway between 0 and 3.3V.

dosipod, 2022-01-23:
Thank you for the explanation , I just noticed that the schematics on the wiki are provided as an example only which makes a big difference in my approach    . While the circuits I made are  simpler I thought  they are  inferior to the wiki  and  I  hope no dragons are involved and that this is safe  . I still have a fused circuit for the leds so  if anything it would be some magic smoke  

The  only aim at the moment is to  help the guys on the forum and discord achieve sound reactive functions and I was not comfortable advising them to setup line-in   other then the way mentioned    in the wiki  . But seeing you are okay with that if it works I could now ask more of them  to try it

netmindz, 2022-02-24:
Does your amp support sending to Bluetooth?

softhack007, 2022-07-14:
You might also want to join up on the Discord to discuss.

https://discord.gg/4CQRmfR