atuline/WLED #159: Why  ESP8266 is going to be depreciate ? Can you just make the 0.13b release version for ESP8266 ?

iyotee, 2022-01-05:
**Take a look at the wiki and FAQ, perhaps your question is already answered!**
[FAQ](https://github.com/Aircoookie/WLED/wiki/FAQ)

**Please consider asking your question on the WLED forum or Discord**
[Forum](https://wled.discourse.group/)
[Discord](https://discord.gg/KuqP7NE)
[What to post where?](https://github.com/Aircoookie/WLED/issues/658)

**If you do not like to use these platforms, delete this template and ask away!**
Please keep in mind though that the issue section is generally not the preferred place for general questions.

FHeilmann, 2022-01-06:
2 Reasons:

- The ESP8266 was never powerful enough to handle WLEDs effects AND the Fast Fourier transformation (FFT) that was required for the more sophisticated effects of the SR fork.
- Many variants of the ESP8266 lack the required amount of flash space for the SR fork

Due to these reasons, the ESP8266 is being phased out of the SR fork.

iyotee, 2022-01-07:
understood. Thanks

dosipod, 2022-02-08:
Hi , 

SR 0.13.B3 for esp8266 is available here  https://wled-install.github.io/  also there is SR 0.13.B4 if you look in discord  but  i haven't tried it . As the guys mentioned esp8266 is not optimal for SR  so expect issues and bottlenecks .
Good luck

netmindz, 2022-02-24:
We should change the description of the project on GitHub to say ESP32 as it still says ESP8266

softhack007, 2022-06-18:
README states that the ESP8266 branch is no longer receiving regular updates. Reasons explained in https://github.com/atuline/WLED/issues/159#issuecomment-1006640255 

Currently you need an ESP32 to run WLED-SR. Technicially the 8266 is not fast enough for our audio processing code, so it's not so easy to 'release th latest dev for esp8266'.

closed 2022-06-18 by softhack007

roboraptor, 2022-06-18:
Im sorry to bring it up here (may be new thread) but, 
Is it possible to have one WLED-SR ESP32 and then just sync the realtime reacting effect through the WLED sync groups to other WLED devices ?

softhack007, 2022-06-18:
Hi @roboraptor, You might also want to join up on the Discord to discuss (https://discord.gg/KuqP7NE), where the community (including developers) might be able to help you.

netmindz, 2022-06-25:
> Im sorry to bring it up here (may be new thread) but, Is it possible to have one WLED-SR ESP32 and then just sync the realtime reacting effect through the WLED sync groups to other (even ESP8266) WLED devices ?

Yeah this is possible, I started hacking this together. It's sadly not that straight forward as the esp8266 branch has diverged too much from the ESP32 branch so which pattern is which doesn't sync 

To do properly would require starting from the main dev branch, then only excluding the FFT processing if you are in ESP8266. However as mentioned you are still going to struggle with the space issue with many ESP8266 based boards so upgrade really is the best option