atuline/WLED #160: How to solve it. Mine is 4M flash ESP32. Do I need to modify the partition table?

zx804262156, 2022-01-09:
bin_rename_copy([".pio\build\esp32dev\firmware.bin"], [".pio\build\esp32dev\firmware.elf"])
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
Error: The program size (1355498 bytes) is greater than maximum allowed (1310720 bytes)
RAM:   [===       ]  34.3% (used 112424 bytes from 327680 bytes)
*** [checkprogsize] Explicit exit, status 1
Flash: [==========]  103.4% (used 1355498 bytes from 1310720 bytes)
bin_gzip([".pio\build\esp32dev\firmware.bin"], [".pio\build\esp32dev\firmware.elf"])
How to solve it. Mine is 4M flash ESP32. Do I need to modify the partition table?

atuline, 2022-01-09:
You need to select a different enviornment. In my case, I use env:soundReactive_esp32dev, which has the proper definitions. So yes, the partition table needs to be updated.

softhack007, 2022-06-18:
Closed, as the reporting user seems to be happy with the answer - at least there was no new response since 6 months.

Please use discord to ask questions and request support from the community.

closed 2022-06-18 by softhack007