atuline/WLED #161: MQTT disabled in 13b6 ?

DarthWeber, 2022-01-11:
**Describe the bug**
in FW13b6 i can not activate MQTT on my esp32 - it is disabled, everything blank and even after filling out everything including the port, broker, topics and activating it, it is after reboot disabled and everything empty again.

**To Reproduce**
fresh installation with "esptool.exe --port YOUR_SERIAL_PORT -b 460800 --before default_reset --after hard_reset --chip esp32 write_flash --flash_mode dio --flash_size detect 0x00001000 bootloader_dio_40m.bin 0x00008000 partition_4mb.bin 0x00010000 fw.bin"

used the esp32 binary v13b6 published here. The issue does not occur with the sound reactive version from sergj (wled shield mit dallas usermod)

DarthWeber, 2022-02-05:
tried today with https://install.wled.me/ - but the problem is the same: I can not activate MQTT. Can somebody confirm or did i do something wrong ?

atuline, 2022-02-05:
As documented in the Wiki at: https://github.com/atuline/WLED/wiki/My--device--is-not-working

these services are disabled in SR by default. You will need to enable them in the source code and re-compile. We do not provide support for those services.

softhack007, 2022-07-14:
MQTT is not active in our standard "soundreactive" builds. You need to build your own binary, without `-D WLED_DISABLE_MQTT` in your `build_flags`. However we have disabled this feature on purpose in SR WLED, so it could be that you experience slow-down of SR effects when you enable MQTT.

Closed as the problem described is intended behaviour.
Please join our Discord where we have a #support channel.

closed 2022-07-14 by softhack007